jschoch/ESPels #4: Hardware details?

kingjamez, 2021-01-11:
Any photos or description of how you mounted the motor to your lead screw and what motor / driver you are using?

I have the same lathe, and have enough spare steppers/servos, ESP boards to build this. Would like to have similar gearing to your setup so that we can share code with no modifications.

jschoch, 2021-01-12:
I've merged in my latest code to the master branch and updated the readme.  I need to update the instructions for setting the leadscrew, microstepping, encoder resolution.  I also need to make the display completely optional, right now it uses the I2C pins.

The display pins are in display.cpp line 10
The rest of the settings are inMachine.cpp starting at line 5
```
float lead_screw_pitch = 2.0;
int microsteps = 16;
//  400 for a .9 degree stepper
int native_steps = 200;
```
Finally the encoder resolution is setup for CPR in Encoder.cpp line 8

int spindle_encoder_resolution=2400 ;

My motor is literally just clamped to my bench so it would be good to design a proper mount.  

tell me a bit about your setup, what encoder are you using, what stepper/stepper driver, what pitch is your leadscrew?

I'm also interested in feedback about the display and buttons.  My plan is to next integrate my haptic input controller but I don't know how well it will work or how many folks want one.  I plan to make it optional. Is this something you are interested in or do you want plain buttons.  If buttons what buttons do you think you'd want?  

It is very usable without buttons at this point and I think it really only would need 4 buttons, "left", "right", "modify pitch", and "rapid" where left would start feeding z-, right would start feeding Z+, holding down modify pitch and left/right would allow you to change the pitch, and holding down rapid with left/right would move in rapid mode.

jschoch, 2021-01-12:
here is the encoder, it is a few pieces of angle.  The gear can be found here:

https://www.thingiverse.com/thing:4720308

[image]

jschoch, 2021-01-12:
The motor is using an HTD 3M belt and is just clamped to the bench.  It could be mounted to the banjo but I'm not sure how rigid that would be, the spindle motor plate could be used but it would require disassembly and machining.  Perhaps the spindle motor mounting bolts could be re-used with a thick plate extending to the leadscrew motor.  Alternatively it could just be mounted to the bench with a bracket.

[image]

jschoch, 2021-01-12:
Finally the network: you need to configure your  ssid and password.  You can hardcode this or make a secret.h file like so

<path>/secret.h  
```
#pragma once

static const char* ssid = "my network name here"
static const char* password = "very secure wifi password"

```

then replace line 8 of web.cpp with the path to that file, best to put it outside of the project file so it doesn't accidentally get pulled into git.

jschoch, 2021-01-12:
I'm going to be making the settings persist on the esp32 and configurable via the webUI so you can change your microstepping etc, it needs to be re-compiled currently if you change those.  

I can also make it fire up with it's own network on first boot and you could configure everything by connecting to its network and then having it reboot and connecting to your network.  

Once it is connected to your network you can update it via OTA.

kingjamez, 2021-01-12:
This is super helpful, thank you! I just bought my MX-210V lathe on Sunday so I'm still in the process of disassembly / cleaning / tuning. I don't even know what my leadscrew pitch is, but I believe it's metric and standard for whatever the *210 lathes use.
I've got everything here save for a rotary encoder. It looks like yours is 600ppr with quadrature output, or is it 2400ppr with quadrature output?

I've got a bunch of steppers, I'm considering mounting the motor at the end of the bed rather than next to the rest of the gears. Will investigate as I finish the teardown / rebuild.

jschoch, 2021-01-12:
the leadscrew pitch should be 2mm but you should check it with a thread gauge.    The encoder is 600PPR which is 2400 CPR  quadrature.

kingjamez, 2021-01-12:
Thanks, I've got the encoder in my shopping cart. Working on filling it up with any other pieces I need to build the electronics side, watching your haptic video again to see if I can build one of those options as well as the push buttons.

jschoch, 2021-01-12:
the haptic controller may work with a nema17, I haven't quite got that working yet.

kingjamez, 2021-01-13:
Did you really print the gear out TPU? I was just about to start printing in in PETG but TPU seems like a very specific choice.

jschoch, 2021-01-13:
i used petg, not sure where that came from

kingjamez, 2021-01-13:
Cool, just printed it in PETG. I'm going to work on a 3D printed bracket for the encoder and will put it here if it works well.

kingjamez, 2021-01-14:
I bought this 4 way switch.
https://www.amazon.com/gp/product/B01IP07824/ref=ppx_yo_dt_b_asin_title_o00_s02?ie=UTF8&psc=1
[image]

It'll be here on Saturday. I'll get to work on building the circuit and an enclosure for it when it arrives. I plan to put it inside the same cavity with the speed control, it's cavernous in there and will be a good place for the joystick.

Can you envision any other hard buttons? I'm thinking of adding a place in the design of my 3D printed faceplate. Something like "set zero" , "set start", or similar for commonly used features

jschoch, 2021-01-16:
my power's been out for the last 4 days, sorry for the delay.

For the "hard buttons" i want to try to limit it to things you really want to be able to press without looking.  I would put set zero in the the webGUI since you are not using it often.  "Rapid", or "change pitch" seem like things you would use often.

This all said it doesn't matter all that much if it is easy to add/remove buttons and maybe it is best to think of a good way you can make the physical controls the way you want as configurable module.

kingjamez, 2021-01-18:
Yikes, 4 days is a long time without power! Glad you got it back.

My encoder arrived and it has a smaller shaft than the gear on Thingiverse (mine is 6mm).

What mod is the gear? I will just design a new one in Fusion. I'll parameterize it so that it can handle different shaft diameters and post it here.

I'm going to tackle the motor mount soon. I might put it on the far end of the lathe if it something doesn't strike me as easy near the banjo. The only problem with putting it on the far end would be running the wires that far. 

I'm building the other electronics now. Am probably going to go start with a TMC5160 and a NEMA23 , running at 24VDC / 1.5 amps to start. Will be interested to see if that's enough power as the TMC5160 won't require much heat-sinking at that power.

Do I need any home / stop switches?

jschoch, 2021-01-18:
the gear is mod1

No homing switches needed 

The stepper driver should be ok generally but maybe not for large threads.

kingjamez, 2021-01-18:
Am I going nuts or is the Thingiverse gear 54 tooth? I counted 4 times and got that each time. As far as I can tell all of these 210 sized lathes use a 56 tooth gear on the spindle. I know this can be accounted for in software, but shouldn't it be a 56 tooth gear for a 1:1 ratio between spindle and encoder?

jschoch, 2021-01-18:
I just checked my machine and gear.  Wow, not sure how I got that wrong.
It is 56 and it should be 1:1 but I have a 54 tooth gear on the encoder and
now I am completely baffled as to how it was working in my tests!

Sorry for the confusion.

kingjamez, 2021-01-19:
Here is a link to the gear I made / printed today: 

https://www.thingiverse.com/thing:4728794

kingjamez, 2021-01-21:
Sorry for the long reply time. Work has kept me too busy to advance much further. I haven't programmed the ESP yet as I'm getting the encoder and motor mounting done first. Hopefully I'll have the mechanical side done this weekend and can start on the electronics soon after.

jschoch, 2021-01-21:
I'm using a lolin d-duino 32 which has an integrated OLED screen.  The only pins being used right now are the step/dir and I2C pins for the display which I can try to make optional.  The step/dir pins only need to be able to be configured for outputs so any pins that are available for that should be ok to use.  I use platform.io to compile/flash which should be able to pull in the dependencies.

Regarding hardware inputs like your switch we should start a separate issue to discuss what you want to use and how we can make those inputs modular and configurable.

The frontend is just run with nodejs.  You should be able to run 

```
npm install
HOST=0.0.0.0 npm start
```

This will start a web server on the machine and you can connect to that ip via the url

```
http://localhost:3000/

or 

http://<your IP here>:3000/
```

If you want I can compile the page and put it up on s3, no server is required for the compiled version.  It is just react and javascript to interact with the websocket of the esp32.  

Once you have it up and running you can go to the network tab and put in the esp32 ip and websocket name (also should be configurable)  the websocket name is [here](https://github.com/jschoch/ESPels/blob/master/src/src/web.cpp#L28)

`ws://192.168.1.93/test`

I've also enabled MDNS but it is a bit finnicky, if it works that would be:

`ws://mydeskels.local/test`

the MDNS name is defined [here](https://github.com/jschoch/ESPels/blob/master/src/src/web.cpp#L370)

kingjamez, 2021-01-21:
Ok, some progress but also some issues.
1. Your example of secret.h above needs semi-colons at the end of the lines.
2. I'm doing my development on a Raspberry Pi 4. I didn't realize that an external server was required. I install nodejs and nom on it. In order to get "nom install" to work, I first had to run "npm init". I didn't know what options to use there so I just went with the defaults.
3. Now when I run "host:0.0.0.0 nom start" I get an error saying "missing script: start"
4. I'm running a Heltec WiFi Kit 32, it's OLED display is on GPIO 4 and 15. Despite changing from GPIO 5 to GPIO15 in the display.cpp, I still don't get anything on the display. Still working on that.
5. I do have the ESP32 attaching to my local network. It's getting DHCP, so I'm not quite sure what the network information in web.cpp (lines 22-25) are supposed to be set to. There is no comment describing those settings. Are they supposed to be the external web server running nodejs? 

I've worked on this a lot today and I think just a few graphics / schematics / description of the architecture of the system would be super helpful. I'm trying to figure it out on my own and if I do, I'll try to make the docs I wish I had.

jschoch, 2021-01-21:
1. fixed
2. An external server isn't required, but I have not posted a compiled version yet since there is plenty of dev yet to do.
3. did you run `npm install` in the cloned directory with the package.json ?  it needs to pull down react which has the start script.  I don't think you need to run npm init at all.
4. it looks like it is a ssd1306 which is the same as the one I'm using.  The i2c address may be different but the display isn't being used for much so you may not want to bother with that yet.
5. they are not used, so don't worry about them.  If you have serial I think it spits out the IP address it acquires.  You connect to that ip in the network tab of the frontend.

it is pre-alpha and "free" so I guess it is what it is.   It will help to document all this for anyone else who wants to use the code but "free" is only relative.  I've not yet invested in documentation because it is pre-alpha and I may make big changes.    There are lots of details to work out like the bidirecitonal haptic interface, making the buttons configurable/modular, and how best to use a display and to have it support multiple display types.

jschoch, 2021-01-21:
I put the frontend up on s3

[http://espels.s3-us-west-2.amazonaws.com/index.html]

kingjamez, 2021-01-22:
Thanks for the help.
Ah.. I didn't see the separate github page for the front end. It's much more clear now.... 

Thanks for hosting it on S3. That's helpful.

-Jim

jschoch, 2021-01-24:
sounds like progress.  Let me know if you have trouble getting it configured.  The encoder direction and motor direction are not yet runtime configurable but that is a planned feature.  Also in the github version of the frontend the only two modes are "startup" which does nothing, and "slave jog ready" which does the moves.  The "free jog" mode did not make it back into the refactored version.  Looking for feedback on what modes are needed.  On my list are: virtual indexer displaying degrees, and a turn counter (i was making a copper coil and realized it could count turns for me).

jschoch, 2021-01-29:
did you get it up and running?

kingjamez, 2021-02-03:
I did get the ESP32 loaded and have it working on a breadboard. I haven't had much free time over the last week so didn't get much further than that. 

I think my plan going forward is to continue with the ESP32 as a motion controller but will use the mechanical bits from this project: 

https://github.com/kachurovskiy/nanoels 

He's done a lot of good work but I don't think the Arduino Nano is a good choice for microcontroller. 

I really like his 3D printed parts though. Especially the improved bearing holders and even a stepper mount for motorizing the cross slide in his Thingiverse account. 

I'd bet it wouldn't be too hard to use some of what what he's done to run on the vastly superior ESP32 / web platform. 

I have decided to go with the TMC 2160 for the stepper driver. It provides all of the super smooth motion from Trinamic chips at up to 4.3 amps. It talks normal step / direction and has dip switches for setup. It should integrate seamlessly with your code. It also has cool features like stall detection and sensorless homing if one day we integrate SPI comms to the stepper controller.

jschoch, 2021-02-03:
that all sounds good but just keep in mind i have currently have no acceleration code.  At low Z speeds this may not be an issue, and if you are starting the spindle before you issue a move it will also likely never be an issue since the lathe spindle needs time to spin up and this adds natural acceleration to the stepper however if you are running the spindle quickly and trying something crazy like 2mm thread pitch then it may stall.  I'm using a CL57Y closed loop stepper and this will compensate somewhat.  The trinamic drivers are good but they are very small and they simply don't have the thermal handling to handle big loads.  This may never be an issue on small threads but there is a guy on hobbymachinist.com who had to get a bigger driver for his ELS when cutting big threads and the loads can be surprising.  

So net net, i haven't added the acceleration code because i've not yet had an issue with the stepper stalling.  I can add it but it is lower on the priority until there is an actual issue.  If the stepper cannot keep up and it needs an acceleration curve there will likely be portions of a thread that would not be accurate.  This is accountable for with a leadin and end thread relief.  

Let me know when you make more progress, i've been using mine quite a bit as is and nothing has yet annoyed me enough to start coding again.

kingjamez, 2021-02-04:
This is not a standard TMC driver. It has external MOSFETS and is designed
to drive big motors. Fortunately it will report missed steps similarly to a
servo. I think it’ll be fine. Acceleration seems hard with syncing to the
spindle... interesting problem.

jschoch, 2021-02-19:
Hey Jim,

Hoping you've made some progress.  I did some threading recently and was noodling around with how to calculate passes and infeeds.  Are you to the point you can help me work out the details?

jschoch, 2021-03-24:
One more ping before I close out this issue? 

I've been using it regularly and it has been well behaved, unfortunately that also means i have no need to dive back into the code so I've made no updates.

kingjamez, 2021-03-24:
Hi, Sorry I've been out of touch. I'm in the middle of some significant health issues and can't work in the shop right now. Go ahead and close out and as I get back to health I'll pick this back up.

jschoch, 2021-03-26:
sorry to hear that, hope you feel better soon!

closed 2021-03-26 by jschoch