jschoch/ESPels #12: clean up deps

jschoch, 2023-03-01:
lib_deps = 
	REMOVE squix78/ESP8266_SSD1306@^4.1.0
	???? someone may want buttons like an estop!  thomasfredericks/Bounce2@^2.55
	REMOVE most likely bricofoy/YASM@^1.0.4
	Don't think I"m using  janelia-arduino/Vector@^1.2.0
 
Keep and or update

	me-no-dev/ESP Async WebServer@^1.2.3
	bblanchon/ArduinoJson@^6.17.2
	bblanchon/StreamUtils@^1.6.1

digiexchris, 2023-03-01:
> I removed Bounce2 here, is there a better way to do this @digiexchris
> <https://github.com/digiexchris>?
>
> https://github.com/jschoch/ESPels/tree/optional_deps

That's a pretty common way of doing things in the 3d printer world, but
they suffer from a very complicated config.h for less technical users and
it causes a lot of questions. It's fine for now though.

I actually was thinking of plugging a second esp in over i2c and doing
buttons and stuff over that, through a layer that plugs in to the web code.
Separates button logic into something that has to make less assumptions
about a user's hardware setup.

I've been keeping notes, I'll create a few issues with diagrams with
thoughts.

There is a platform io config in the repo for unit tests but I can't find
any tests other than the example, is that something that concerns you?

digiexchris, 2023-03-01:
hmm, I thought you were using yasm?

jschoch, 2023-03-01:
>   There is a platform io config in the repo for unit tests but I can't find any tests other than the example, is that something that concerns you?

I'd love to learn a good way to do testing with MCUs, is there an emulator for esp32?  If not i'm open to adding some tests but I simply don't know how best to setup testing for this type of thing.

Regarding YASM, it is being used but only barely.

digiexchris, 2023-03-01:
I was about to look into it. I'm pretty sure the idea is you use an ESP without needing any other hardware attached, and it'll execute your test code on it and return the results. I've got a few $3 ESP's around, and I bet for our purose, an ESP8266 would be fine. Doesn't need to be multi-threaded for tests. I'm making a lot of assumptions here but it's something I want to know about so I'll do a little more poking around and try it out. I have in-circuit debugging going on with mine through JTAG so I can step through it and figure it out.

digiexchris, 2023-03-01:
can also make the code under test largely independant from the esp framework and run the tests with g++ or gcc. One of the beauties of converting to an object oriented architecture like c++.