jschoch/ESPels #17: Debug feature: Have stepper move in an exact number of steps in a particular direction

digiexchris, 2023-03-02:
Considering I (and others) are going to try to do threading operations in the opposite measurement system than the leadscrew is (eg. an SAE leadscrew doing metric threads) it would be nice to have a way to experimentally measure the actual error given with a certain distance. I'd like a box on the debug screen with the number of steps (microsteps in this case, since that has torque implications too) to move, and a button for making it go in a certain direction. That way I can pre-calculate the number of steps required to make the carriage move the length of the bed, and use my DRO to see how close it came. I want this instead of distance, because I want to be able to separate any math errors coming from my converting my leadscrew to metric in the firmware (floating point rounding errors, accumulating errors, etc) from the results. Might be an easy frontend change?

jschoch, 2023-03-02:
you can do this now by using the 'virtual encoder', I just need to add a text entry for the value.  Then we can compare the steps moved (available in the config screen), the measurement, and the estimated distance.  

you can also do the calculations and compare the distances, pulses, and steps with jogging a given distance.  you'll notice the "DRO" will show the error (it has more precision than displayed, I think it is currently limited to 3 or 4 decimal places).

If you really need steps (not sure that helps since this is all encoder pulse interrupt driven) we can leave this here.  If not we can move it to the elsFrontened repo.

digiexchris, 2023-03-02:
nope you've basically solved it for me. we can move it there.

closed 2023-03-02 by digiexchris