jschoch/ESPels #22: backlash compensation

jschoch, 2023-03-03:
consider a config parameter and some movement to ensure that backlash is removed before trying to sync move

jschoch, 2023-03-11:
Need to flush this out

consider moving to shoulder, only one direction of backlash comp is needed.  
how do you set this up clearly?  
How do you deal with backlash for  Z- || Z+ || both
best usecase might be threading where we are moving back and forth lots, however this is only relevant when you don't have space for extra lead in, but there isn't really a fix for that ( example: tailstock prevents leadin).

digiexchris, 2023-03-13:
With the acceleration Lib this becomes easy. A nieve move distance on any direction change pre-set in config. I think it should be any direction change regardless of direction. I currently fight backlash in setting up the movesync distance.