jschoch/ESPels #25: development mode esp32 memory and compile time bug safety features

digiexchris, 2023-03-07:
In order to ease debugging and warning of unexpected behaviour, I think turning these on in the sdkconfig will help.

Heap Poisoning
Stack Smashing Protection Mode
UBSAN  should help too with extra compile time checks and is probably generally useful for preventing bugs including divide by zero bugs.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/fatal-errors.html

There is performance implications for these, but might not be much. We can always turn them off for a full release.

digiexchris, 2023-03-08:
yeah it seems it's not a pio problem, it's the arduino framework problem. Yes I'd like esp-idf instead. I've used the arduino component in an Espressif project and it works well.

The other upside is then you can use any ide or toolchain you want, since it's just cmake. It's not bound to pio.

jschoch, 2024-01-16:
renaming this, espressif and platformIO seem to have ended partnership and there is no offical support for IDF 5.0.1+

Also, RMT api used is depricated and needs to be updated but platformio does not yet support the new API.