jschoch/ESPels #26: LED status

jschoch, 2023-03-08:
considering 2 setups:

simple:  CONTROLLER_ERROR_LED, WIFI_STATUS_LED

advanced:

```
ws2812_CONTROLLER_STATE  {
  startup_mode: yellow,
  running_mode: orange;
  conjestion_mode: blue;
 ready_mode: green;
  errors: white;
  halt: red;
}
ws2812_WIFI_STATE {
  congestion_mode: white;  // this is when the websocket queue is getting full
  bad_signal_mode: yellow;
  good: green;
 not_connected_to_wifi: red
  no_clients_connected: blue;
}
```

jschoch, 2023-03-11:
basic 2 led timer is in, need to test wifi states.

for error codes, if it reboots you'll likely never see them unless you pause the reboot.  not sure what to do with this case.

digiexchris, 2023-03-14:
I setup the wifi led, it just blinks, I don't get any additional info from it. it may be connecting fast enough that I won't see it, so that's no big deal. Might be nice for it to change pattern if RSSI goes under a certain value?

You could use an error handler, and basically shut down all tasks and interrupts to display the blink code. Then hit the reset button to reboot. That way, if it runs into a weird error, you could connect a debugger and take a look

jschoch, 2023-03-14:
open to PR's on the blink codes, was thinking that when a client is connected it could change the pattern.

Agree on the error handler, open to ideas on what blink codes we can use.

https://github.com/jschoch/ESPels/blob/dev/src/src/led.cpp

defazioa, 2023-12-05:
What you're describing here is an _Andon_ bollard. It's a very good idea, but should be capable of interpretation at a glance. There's little time for counting blinks around machine tools. I like your ideas, I have looked at a lot of them here and elsewhere, and I have Jon Bryan's Mega2560 based "Atomic ELS" installed on my lathe now. I would prefer to install and evaluate yours if I can figure out how. I may have some input from an end user point of view I'll put here if there's a relevant problem. Thanks for all your work, it's very impressive.

jschoch, 2023-12-05:
> What you're describing here is an _Andon_ bollard. It's a very good idea, but should be capable of interpretation at a glance. There's little time for counting blinks around machine tools. I like [...]

feel free to start a new issue and document any issues or questions you have.