jschoch/ESPels #29: Current bugs from testing 0.0.3

digiexchris, 2023-03-12:
Bounce mode doesn't seem to work, or at least maybe a bug moving back and forth between unit types. When I did get it to move in bounce it appeared to be moving in metric with an inch value. It was also switching the front-end into a bounce mode enum with a blank page blow it, I had to switch back to movesync and then bounce tab to cancel it  further attempts at bounce did no moves.

In move sync mode  the left rapid moves correctly to the left. The right rapid moves incorrectly to the left instead of the right.

Unit conversions still seem broken. Move distance converts ok, but the pitches sometimes don't convert properly when switching units back and forth. Not sure what causes it. Sometimes it works, sometimes it doesn't and I end up with a very long slow pitch like 0.000000577853

Good news is there only seems to be about 0.06 mm error over 25mm. I cut a 20tpi thread that was 3/8" long that fit well so it at least satisfies my short thread needs.

jschoch, 2023-03-12:
Thanks for the report, 

Did you use the latest dev?  This commit fixed the bounce display mode issue https://github.com/jschoch/espELSfrontend/commit/5602be88ecfbc56ede1dcde468e8aa6c5ad6368f

jschoch, 2023-03-14:
should be resolved

closed 2023-03-14 by jschoch