jschoch/ESPels #37: Configuration in progress

defazioa, 2023-12-06:
Here are a few photos and some questions:
[image]
[image]
[image]
[image]
As you can see, I got lucky and was able to shoehorn the servo in the unused space under the headstock on this particular lathe. The levers on the front are three position switches; one on the right is maintained and the left one is momentary. I would like the maintained one to determine direction. I was also thinking that the existing hall sensor that was used by the factory tachometer could be put to work as an index pulse to synchronize the carriage

defazioa, 2023-12-06:
The display is Nextion, the ELS running on the machine now is from **repo:ccomito1223/ELS-Jon** found on this site. It runs on this Mega2560:
[image]
It works, but won't change directions. Threading is impossible, and there simply doesn't seem to be enough processor or clock headroom to do what it's doing.

defazioa, 2023-12-08:
I have managed to flash the code and land all the wires on the ESP32. Unfortunately, I have been unable to make the frontend work so I can't test it. I'm out of my depth here, I think we could help each other, but like I said, I don't know what I'm doing. I am poorly self taught I guess you could say. I gave it two solid evenings and I'll keep trying, but my confidence is shaken. Sorry.

jschoch, 2023-12-08:
odd, i can't reproduce this, but when I first checked out the s3 link it sent me to https, but you have to be regular http because the esp32 doesn't doe TLS and it will make a cors error if you try to go from https source to http websocket. 

Easy way to see this is via the browser's developer tools, the page is blank and the console says the following

[image]

the url should be  "http://espels.s3-us-west-2.amazonaws.com/release_0_0_3/index.html"  not "https://espels.s3-us-west-2.amazonaws.com/release_0_0_3/index.html".  I need to load the latest release since i'm running an expirimental branch and there are some CORS issues with SSE that don't exist in the latest release and this forces me to run it from npm locally.

defazioa, 2023-12-08:
It's not an S3, it's a node mcu wroom piece that I have a couple of. I have others which are s3's but they're minis and don't fit the breakout I have. The frontend issue is somewhat like you describe, I tried to run it from vs code as well as the node js directory but no dice. It's important that you know that our skills might be exactly reversed; I am a good toolmaker and not a moron but some of this system stuff has me stumped. I started on a nextion project as a stop gap measure because I already have the display installed anyway.

defazioa, 2023-12-08:
Elecrow also sells a display that uses an esp32 as it's mcu with a few it's. I wonder.

defazioa, 2023-12-08:
That was meant to be IO's. I also don't see what I expected on the serial monitor. It appears to be reporting the die temperature every couple seconds. In degrees Fahrenheit oddly.

jschoch, 2023-12-09:
can you paste in this issue the output from the serial console?  You can use markdown to format it, here is a guide

https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet

Can you also paste in here your config.ini, please redact your wifi password.

>It's not an S3, it's a node mcu wroom piece that I have a couple of. I have others which are s3's but they're minis and don't fit the breakout I have. 

when I say S3, I mean the url I provided.  it is hosted on AWS's S3 storage platform.  I see now that it is easily confused with esp32-S3.  I've not tested S3, only the older dual core esp32s.

The web frontend is just javascript, it only needs to be able to get to the ELS's IP via a http websocket.  so no "server" is really needed to interact with, just load the html, css, and javascript.  This is all bundled and available here:  "http://espels.s3-us-west-2.amazonaws.com/release_0_0_3/index.html"   You can also run it locally from any web server or local file but it needs to be "bundled",  You need to setup react and npm and node to run it in development mode.  Easiest just to run it via the link above, just ensure the prefix is "http" and not "https"

Regarding the frontend, if you use the http (not https) link above you should be able to load the UI.  You then tell it what IP address your esp32 is via the network tab.  You should see the IP in the serial console when the esp32 boots.  Here you can see mine was set to 192.168.1.87 in the webUI.  Note the ws:// and the trailing /els path.  these are both needed.  

[image]

To get the browser developer console press the keys <ctrl><shift><i>  all at once with the browser in focus.

The browser console should look like this:

[image]

your serial output should look something like: 

```
--- Terminal on COM5 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
1 (POWERON_Rrecalculated denominator: 4800Set ELS Factor pitch: 0.100000 nom: 80 den: 4800
Set ELS Factor pitch: 0.100000 nom: 80 den: 4800
Setting up WiFi
reConnecting to Wi-Fi...
................................................................Connected to Wi-Fi.
Connected. IP=192.168.1.87
HTTP websocket server started
[  6616][I][ArduinoOTA.cpp:141] begin(): OTA server at: esp32-30c6f72f9640.local:3232
NV Config?
1
Loaded Configuration com version 0.0.3 lead screw pitch: 1.333333
Loaded this NvConfig doc{
  "i": 1,
  "lead_screw_pitch": 1.333333333,
  "spindle_encoder_resolution": 2400,
  "microsteps": 4,
  "EA": 34,
  "EB": 35,
  "ZP": 26,
  "ZD": 25,
  "motor_steps": 800, 
  "native_steps": 200,
  "vsn": "0.0.4",
  "t": "nvConfig"
}recalculated denominator: 3200Set ELS Factor pitch: 0.100000 nom: 80 den: 3200
Using External PULLUP/PULLDOWN
setup done
 0 0 -57 Websocket client connection received id: 1
helo received, sending nvconfig
 0 0 -56  0 0 -56 ^ 0 0 -55  0 0 -57 ^ 0 0 -56

```

the numbers it spits out are some old debugging stuff, it is:

<run mode> <position> <wifi strength>

so on first boot, 0 0 -56 where the first zero is run mode 0, the second 0 the stepper position, and finally -56 RSSI the wifi strength in db

defazioa, 2023-12-09:
Good news, frontend is up and connected. Bad news is that nothing works, which is kind of par for the course on these things. I don't screw movement with venc or RPM in the app, I'll get the scope out and make sure it's clean going in. I probably need to make some adjustments to the resistor values, the arduino was 5vdc. My lathe has a feed lever and power cross feed along with the half nut. My lead screw has a keyway cut it's entire length to facilitate this. The intent is to preserve the lead screw threads for thread cutting. So the question is will the leadscrew stay in sync with the spindle at all times so you can use the half nut and thread dial? As I said, I have the old hall sensor from the original tach still installed, could you use a pulse from it to reset the sync cycle every rotation? With an interrupt I mean. If that's a dumb question, remember I don't know nearly what you do about this stuff. I didn't make any changes to anything but the config file, most of the trouble was a new workstation that was hijacked for a short time by one drive so the directories weren't where they were supposed to be. Sorry to be so long winded.

jschoch, 2023-12-09:
> Good news, frontend is up and connected. Bad news is that nothing works, which is kind of par for the course on these things. I don't screw movement with venc or RPM in the app, I'll get the scope [...]

the esp32 is 3.3v.  what stepper controller are you using?
Regarding the encoder, what type is it? do you have pullups connected?  If not you have to update your config.ini.

> My lathe has a feed lever and power cross feed along with the half nut. My lead screw has a keyway cut it's entire length to facilitate this. The intent is to preserve the lead screw threads for [...]

as soon as you disengage the half nut you lose sync.  In practice you'll still use it but only when it is ok to loose sync.

> As I said, I have the old hall sensor from the original tach still installed, could you use a pulse from it to reset the sync cycle every rotation? With an interrupt I mean. If that's a dumb question, remember I don't know nearly what you do about this stuff.

I have not yet tested an encoder with an index pulse, but I do have one now and was planning on playing around with this.  In several years of using it I haven't needed the index pulse.

> I didn't make any changes to anything but the config file, most of the trouble was a new workstation that was hijacked for a short time by one drive so the directories weren't where they were supposed to be. Sorry to be so long winded.

Please paste the config, you have to configure the encoder pins and the step/dir pins to match your setup unless you set them up the same as in the example config.ini

Also, I notice your lathe has a gearbox.  the setup is only for one ratio of spindle to leadscrew and there is no way to let the controller know you changed gears.  This could be added as a feature but right now it is assumed you will always have the same spindle to leadscrew ratio and so you should always have the same gear selected when you are running it.

Finally, how did you try to run it, can you detail it step by step?   in "startup" mode You should be able to see the rev counter when turning the encoder or turning on the virtual encoder.  This also should show up in the debug page with the "Encoder Pulses" field updating

[image]
 
Once we can confirm the encoder is working and has the roughly the right counts we can move on to the stepper.

defazioa, 2023-12-09:
The motor is an integrated "servo" with drive attached, was quiet and happy running the old system. The encoder is bog standard 360 cpr but driven x2 for 720 cpr. The lathe _**had**_ a gearbox that was noisy and obnoxious, what you see now is an empty box with the two rotary switches inside described above. I modified them to use the original handles. When I learn how to modify your code I'll turn them into feed direction and rapid. Here';s the config:
[env:ESPels]
#port that platformio will use to upload over usb
upload_port = COM6

[env:ESPelsOTA]
#port that platformio will use to upload over usb
upload_port = COM6

build_src_flags = 
#machine configuration
	-D HOSTNAME="WBL250Flathe"
    -D LEADSCREW_LEAD=4.0
    -D WIFI_SSID="Verizon_DK67HR-IoT"
    -D WIFI_PASSWORD=""
#motor configuration
	-D Z_STEP_PIN=13
	-D Z_DIR_PIN=12
	-D Z_NATIVE_STEPS_PER_REV=200
    -D Z_MICROSTEPPING=4
#Spindle Encoder pins
	-D EA=25
	-D EB=26
    #choices are: 
    #EXTERN //your hardware handles the pullup or pulldown, for example with a resistor to 3v3 or to gnd
    #INTERN_PULLDOWN //your encoder expects the MCU to pull low, and sends a high for a pulse
    #INTERN_PULLUP //your esp handles the pullup and your encoder pulls it down.
    #For most common quadrature encoders, choose Extern and apply 3.3v to your encoder pins through 1.5k to 4.7k ohm resistors */
	-D ENCODER_PULLUP_TYPE=EXTERN
    #ENCODER_QUADRATURE_MODE should be ON, unless you have a good reason not to. Otherwise your leadscrew will not track reverses.
    #valid values: ON, OFF
	-D ENCODER_QUADRATURE_MODE=ON
	-D ENCODER_RESOLUTION=2880 #this is counts per rev *4 for a quadrature encoder
	-D USESSD1306=0 #use the display or not. Recommend not, it doesn't have much functionality. Set to 1 to enable.
	-D error_led=21
    -D configured=true #ignore this, this just tells the config that you've configured your machine

[env:ota]
#ip address of the esp when connected to your wifi, if you want to upload firmware remotely
upload_port = 192.168.0.187

#[env:ESPelsOTA]
#ip address of the esp when connected to your wifi, if you want to upload firmware remotely
#upload_port = 192.168.1.28

defazioa, 2023-12-09:
Oh, yes there are pullups. I'm going to look at the waveforms now to see if I need to do anything. I think they're 4.7k.

defazioa, 2023-12-09:
[image]
Got good encoder data which diplays RPM and rel position updates in the app, but no pulses or direction logic from the controller on pins 12 and 13. It's a good start, I poked around a little on other pins and found a square wave on pin 32, but it doesn't change with speed. Thanks for your help.

jschoch, 2023-12-09:
Good progress!  

In order to generate steps you need to change the mode in the app. 

[image]

 Select "feed mode" and in that screen click "turn on"

You should then see the DRO update when you move the spindle.  It will look like this

[image]

defazioa, 2023-12-10:
I have pulse and direction signals from the controller but no movement, so I have a problem on my end. After all this the wire management is looking decidedly "prototype" so I'll probably find it here in a minute. The encoder pulse count keeps climbing without resetting, this inhibits jogging and throws an error message. I also need to figure out ota flash, I noticed a problem with my config while scrolling past it here.

defazioa, 2023-12-10:
I had to put in a level shifter to bring the jolts up to something the drive can see, but after doing so the MCU wont connect to wifi. Total mystery. I downloaded a fresh zip file from your repo, changed the config, and tried to compile it. Just like last time it failed to start ninja. I thought this was a result of my inexperience but it could be related to Microsoft requiring admin rights to run powershell on win 11. I got it to work last time but it was tedious and took forever. I'm working on it, sorry for the delay.

jschoch, 2023-12-10:
what level shifter are you using?  That is strange that it would effect wifi.  Wifi is the biggest power draw, maybe your power supply for the esp32 is too small?  should have 1A of 5v power.

regarding Ninja, I don't think i'm using that.  are you using vs code + platformIO extension?  that is what i'm using.

For any build system you shouldn't have to reinstall, you should be able to run a `build clean` if things get wonky.

One thing i noticed was that the new arduino esp32 library in platformio has problems with the deprication of the RMT library 

in platformio.ini i set it to use the older version, 5.4
```
[env:ESPels]
platform = espressif32@5.4
```

here is my build out put dependency graph

```
Processing ESPels (platform: espressif32@5.4; board: esp32dev; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (5.4.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6)
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- FastAccelStepper @ 0.28.4
|-- ESP8266 and ESP32 OLED driver for SSD1306 displays @ 4.4.0
|-- Bounce2 @ 2.71.0
|-- YASM @ 1.0.5
|-- ESP Async WebServer @ 1.2.3+sha.f71e3d4
|-- ArduinoJson @ 6.21.4
|-- StreamUtils @ 1.8.0
|-- ESP32Encoder @ 0.10.2
|-- Ticker @ 2.0.0
|-- ArduinoOTA @ 2.0.0
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
Building in release mode
```

defazioa, 2023-12-10:
This sounds crazy but I built this in the ESP-IDF. No wonder it was so hard to round up dependencies and get it to work. I guess I did something really hard and thought it was my incompetence. Go figure.

jschoch, 2023-12-10:
wow, ummm, wow

I gave that a try once, gave up pretty quickly.... kinda amazed it works at all.

defazioa, 2023-12-10:
I'm still having some issues with the pio build. It seems that the config.ini is not being read. I have only altered the config.ini.template.ini and renamed it. There's no place to paste it. what directory is it supposed to live in?

jschoch, 2023-12-10:
it should be in the same directory as your platformio.ini

defazioa, 2023-12-10:
It is. What does your platform.ini have in it?

jschoch, 2023-12-10:
I tested the release version in this repo, it works fine for me.  you need to update the OTA stuff like IP address but otherwise it should be fine.  The only change is to add the older espressifArduino tag, you can try to change the name of the config.ini, it is specified in the extra_configs option per below.

```
[platformio]
extra_configs = SOME_OTHER_NAME.ini
[env:ESPels]
platform = espressif32@5.4
```

defazioa, 2023-12-10:
I assume your development platform has been optimized for this kind of thing, so maybe that's why I'm struggling. All the errors I get are from the values declared in the config.ini. The file is in the ELS0.0.3 directory. I have deleted and re-downloaded this thing a dozen times and changed it until I couldn't track the changes then reverted, then downloaded again to be sure. It has to be a missing call because it throws an alarm for every single variable in the file.

jschoch, 2023-12-10:
paste in the errors you are seeing.

Please also show the build dependency output, in particular what version do you see in the packages section

```
PACKAGES:
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6)
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
```

defazioa, 2023-12-10:
There you go again... what makes you think I have any idea where to find that? I've made parts that are still working in space. It amazes me that I've invested so many hours and made zero progress since last night. Frustrating.

jschoch, 2023-12-10:
to get to the build menu you have to click on the alien thing on the left

[image]

when you click build under the ESPels env, as below

[image]

then you have the output under "terminal" , there is also a list of tasks to the right, one of them is the build output.   You may have to scroll up to get to the beginning and to see the package section.

[image]

defazioa, 2023-12-10:
Scanning dependencies...
Dependency Graph
|-- FastAccelStepper @ 0.28.4
|-- ESP8266 and ESP32 OLED driver for SSD1306 displays @ 4.4.0
|-- Bounce2 @ 2.71.0
|-- YASM @ 1.0.5
|-- ESP Async WebServer @ 1.2.3+sha.f71e3d4
|-- ArduinoJson @ 6.21.4
|-- StreamUtils @ 1.8.0
|-- ESP32Encoder @ 0.10.2
|-- Ticker @ 2.0.0
|-- ArduinoOTA @ 2.0.0
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
It does not look at the config file. Everything fails. This is from just now, a new download. Here are the parameters if you want to try to compile it on your end. The wifi password will be temporary:

[env:ESPels]
#port that platformio will use to upload over usb
upload_port = COM6

build_src_flags = 
#machine configuration
	'-D HOSTNAME="WBL250FLATHE"'
    -D LEADSCREW_LEAD=2.0
    '-D WIFI_SSID="Verizon_DK67HR-IoT"'
    '-D WIFI_PASSWORD="sighs-dock4-map"'
#motor configuration
	-D Z_STEP_PIN=18
	-D Z_DIR_PIN=19
	-D Z_NATIVE_STEPS_PER_REV=200
    -D Z_MICROSTEPPING=4
#Spindle Encoder pins
	-D EA=25
	-D EB=26
    #choices are: 
    #EXTERN //your hardware handles the pullup or pulldown, for example with a resistor to 3v3 or to gnd
    #INTERN_PULLDOWN //your encoder expects the MCU to pull low, and sends a high for a pulse
    #INTERN_PULLUP //your esp handles the pullup and your encoder pulls it down.
    #For most common quadrature encoders, choose Extern and apply 3.3v to your encoder pins through 1.5k to 4.7k ohm resistors */
	-D ENCODER_PULLUP_TYPE=EXTERN
    #ENCODER_QUADRATURE_MODE should be ON, unless you have a good reason not to. Otherwise your leadscrew will not track reverses.
    #valid values: ON, OFF
	-D ENCODER_QUADRATURE_MODE=ON
	-D ENCODER_RESOLUTION=2880 #this is counts per rev *4 for a quadrature encoder
	-D USESSD1306=0 #use the display or not. Recommend not, it doesn't have much functionality. Set to 1 to enable.
	-D error_led=21
    -D configured=true #ignore this, this just tells the config that you've configured your machine

[env:ota]
#ip address of the esp when connected to your wifi, if you want to upload firmware remotely
upload_port = 192.168.0.187

[env:ESPelsOTA]
#ip address of the esp when connected to your wifi, if you want to upload firmware remotely
upload_port = 192.168.1.28

jschoch, 2023-12-10:
I need the version above the dependencies you pasted, it should be at the top under PACKAGES

`- framework-arduinoespressif32 @ <your version should be here>`

I also have no idea why it is not compiling because I can't see any of the errors.    While you may see lots of errors for the config.ini, it may actually be something else.  The first errors typically are most important and then c++ just vomits out tons of other stuff that may or maynot be related.

defazioa, 2023-12-11:
This is the entire output from the terminal. If it says "packages" I missed it. My platform.ini has a bunch of stuff in it which I replaced with yours containing almost nothing copied from this chat. It made matters far worse so I started adding sections back in. There is also an "ESPels_Master" in the repo which looks different. I must be driving you nuts, It pisses me off that I had the damnmed thing working last night and I can't even change the controller because I'm too dumb to compile the code. Weeping Jesus on the cross.

 *  Executing task: C:\Users\defaz\.platformio\penv\Scripts\platformio.exe run 

Warning! `check_patterns` configuration option in section [env:cppchecker] is deprecated and will be removed in the next release! Please use `check_src_filters` instead
Processing ESPels (platform: espressif32; board: esp32dev; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.4.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- FastAccelStepper @ 0.28.4
|-- ESP8266 and ESP32 OLED driver for SSD1306 displays @ 4.4.0
|-- Bounce2 @ 2.71.0
|-- YASM @ 1.0.5
|-- ESP Async WebServer @ 1.2.3+sha.f71e3d4
|-- ArduinoJson @ 6.21.4
|-- StreamUtils @ 1.8.0
|-- ESP32Encoder @ 0.10.2
|-- Ticker @ 2.0.0
|-- ArduinoOTA @ 2.0.0
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
Building in release mode
Compiling .pio\build\ESPels\src\src\led.cpp.o
Compiling .pio\build\ESPels\src\src\web.cpp.o
Building .pio\build\ESPels\bootloader.bin
Generating partitions .pio\build\ESPels\partitions.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio\build\ESPels\lib72a\FastAccelStepper\FastAccelStepper.cpp.o
Compiling .pio\build\ESPels\lib72a\FastAccelStepper\PoorManFloat.cpp.o
Compiling .pio\build\ESPels\lib72a\FastAccelStepper\RampCalculator.cpp.o
Compiling .pio\build\ESPels\lib72a\FastAccelStepper\RampConstAcceleration.cpp.o
Compiling .pio\build\ESPels\lib72a\FastAccelStepper\RampGenerator.cpp.o
Compiling .pio\build\ESPels\lib72a\FastAccelStepper\StepperISR.cpp.o
Compiling .pio\build\ESPels\lib72a\FastAccelStepper\StepperISR_avr.cpp.o
Compiling .pio\build\ESPels\lib72a\FastAccelStepper\StepperISR_due.cpp.o
Compiling .pio\build\ESPels\lib72a\FastAccelStepper\StepperISR_esp32.cpp.o
Compiling .pio\build\ESPels\lib72a\FastAccelStepper\StepperISR_esp32_mcpwm_pcnt.cpp.o
Compiling .pio\build\ESPels\lib72a\FastAccelStepper\StepperISR_esp32_rmt.cpp.o
Compiling .pio\build\ESPels\lib0a1\ESP8266 and ESP32 OLED driver for SSD1306 displays\OLEDDisplay.cpp.o
Compiling .pio\build\ESPels\lib0a1\ESP8266 and ESP32 OLED driver for SSD1306 displays\OLEDDisplayUi.cpp.o
src/src/led.cpp: In function 'void my_wifi_task(void*)':
src/src/led.cpp:23:22: error: 'wifi_led' was not declared in this scope
         digitalWrite(wifi_led,HIGH);
                      ^~~~~~~~
src/src/led.cpp:23:22: note: suggested alternative: 'wifi_th'
         digitalWrite(wifi_led,HIGH);
                      ^~~~~~~~
                      wifi_th
src/src/led.cpp: In function 'void startErrorTask()':
src/src/led.cpp:32:10: warning: unused variable 'task1Param' [-Wunused-variable]
     char task1Param[12] = "taskParam";
          ^~~~~~~~~~
src/src/led.cpp: In function 'void led_init()':
src/src/led.cpp:58:13: error: 'wifi_led' was not declared in this scope
     pinMode(wifi_led,OUTPUT);
[... 274 lines cut ...]
src/src/web.cpp: In function 'void handleFeed()':
src/src/web.cpp:532:17: warning: format '%f' expects a matching 'double' argument [-Wformat=]
   Serial.printf("\nFeed ccw: %d pitch: %f config pitch %f\n",feeding_ccw,mc.pitch);
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/src/web.cpp: In function 'void init_web()':
src/src/web.cpp:880:17: warning: unused variable 'eventID' [-Wunused-variable]
   WiFiEventId_t eventID = WiFi.onEvent(onWifiDisconnect, WiFiEvent_t::ARDUINO_EVENT_WIFI_STA_DISCONNECTED);
                 ^~~~~~~
src/src/web.cpp: At global scope:
src/src/web.cpp:32:20: warning: 'TAGweb' defined but not used [-Wunused-variable]
 static const char* TAGweb = "Mc";
                    ^~~~~~
In file included from src/src/state.h:10,
                 from src/src/web.cpp:9:
src/src/moveConfig.h:7:20: warning: 'TAGmc' defined but not used [-Wunused-variable]
 static const char* TAGmc = "Mc";
                    ^~~~~
================================================= [FAILED] Took 5.23 seconds =================================================

Processing ota ()
------------------------------------------------------------------------------------------------------------------------------
UndefinedEnvPlatformError: Please specify platform for 'ota' environment

 *  The terminal process "C:\Users\defaz\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

jschoch, 2023-12-11:
the package is in there, it shows PACKAGES:

framework-arduinoespressif32 @ 3.20011.230801 (2.0.11)

This is good!  the error that is catching my eye is:

src/src/led.cpp:23:22: error: 'wifi_led' was not declared in this scope

try adding these to your config.ini with the rest of the -D lines.

```
-D error_led=33
    -D wifi_led=32
```

this is for a blink code, you can change the pins and add an LED which may help you see issues.  Sorry, it doesn't appear it is in the default config.init template.

defazioa, 2023-12-11:
Okay, I apparently killed the first controller. I was finally able to upload to a new one. The pins are in different places, as soon as I wire it up we should be back at last night.

jschoch, 2023-12-11:
Hang in there!  I spent the last 2 days slogging through some tough
troubleshooting with linuxcnc-rio and finally just got it working.

- "It's not that I'm so smart, it's just that I stay with problems longer."
- Albert Einstein
-

defazioa, 2023-12-11:
Here's the status: Finally compiled and uploaded the project. When I ping the address I usually get 2 echoes out of four. I don't know how I'm connected to it but it's likely not peer to peer, but through the router. It did connect with the app briefly, but the readings were spurious. It looks like the reason is the nvconfig. It still has the wrong pins for step and direction according to the debug tab. Why would that be? I also had this error at the very tail end of the build:

Processing ESPelsTest (platform: native)
------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 2 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoFake @ 0.4.0
|-- ArduinoJson @ 6.21.4
Building in release mode
Compiling .pio\build\ESPelsTest\src\main.o
Compiling .pio\build\ESPelsTest\src\src\BounceMode.o
'g++' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\ESPelsTest\src\src\Controls.o
'g++' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\ESPelsTest\src\src\DebugMode.o
'g++' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\ESPelsTest\src\src\Encoder.o
'g++' is not recognized as an internal or external command,
operable program or batch file.
'g++' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\ESPelsTest\src\src\Machine.o
Compiling .pio\build\ESPelsTest\src\src\SlaveMode.o
'g++' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\ESPelsTest\src\src\Stepper.o
'g++' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\ESPelsTest\src\src\display.o
'g++' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\ESPelsTest\src\src\hob.o
'g++' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\ESPelsTest\src\src\led.o
'g++' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\ESPelsTest\src\src\log.o
'g++' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\ESPelsTest\src\src\motion.o
'g++' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\ESPelsTest\src\src\myperfmon.o
'g++' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\ESPelsTest\src\src\neotimer.o
'gcc' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\ESPelsTest\src\src\pcntEncoder.o
*** [.pio\build\ESPelsTest\src\main.o] Error 1
*** [.pio\build\ESPelsTest\src\src\BounceMode.o] Error 1
*** [.pio\build\ESPelsTest\src\src\Controls.o] Error 1
*** [.pio\build\ESPelsTest\src\src\DebugMode.o] Error 1
*** [.pio\build\ESPelsTest\src\src\Encoder.o] Error 1
*** [.pio\build\ESPelsTest\src\src\Machine.o] Error 1
*** [.pio\build\ESPelsTest\src\src\SlaveMode.o] Error 1
*** [.pio\build\ESPelsTest\src\src\Stepper.o] Error 1
*** [.pio\build\ESPelsTest\src\src\display.o] Error 1
*** [.pio\build\ESPelsTest\src\src\hob.o] Error 1
*** [.pio\build\ESPelsTest\src\src\led.o] Error 1
*** [.pio\build\ESPelsTest\src\src\log.o] Error 1
*** [.pio\build\ESPelsTest\src\src\motion.o] Error 1
*** [.pio\build\ESPelsTest\src\src\myperfmon.o] Error 1
'g++' is not recognized as an internal or external command,
operable program or batch file.
*** [.pio\build\ESPelsTest\src\src\neotimer.o] Error 1
'g++' is not recognized as an internal or external command,
operable program or batch file.
*** [.pio\build\ESPelsTest\src\src\pcntEncoder.o] Error 1
================================================= [FAILED] Took 0.80 seconds =================================================

Processing ota ()
------------------------------------------------------------------------------------------------------------------------------
UndefinedEnvPlatformError: Please specify platform for 'ota' environment

 *  The terminal process "C:\Users\defaz\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

jschoch, 2023-12-11:
that looks like the native test mode, you shouldn't need to run that ESPelsTest environment.  It is for development testing and runs some generalized unit tests on your PC which needs a compiler setup for x86.

In the platformio menu (on the left, when you click on the alien icon tingy) each build environment has a toggle (to expand the selection), you only really need to run, build, upload, monitor, and maybe clean in each environment.  You should be using build/upload in the ESPels env, or if you have setup the OTA partitions then you can try to upload over the network via the ESPelsOTA env.  

[image]

You shouldn't run "Build All" in the default environment.  

regarding ICMP ping, i'm not sure how well it should perform but it sounds like you have packet loss over your network.  What RSSI strength do you see in the console (the third number it spits out)?

defazioa, 2023-12-11:
I am at the office right now, do I need to manually change those somewhere when I get home? Please don't take my comments the wrong way, I chose this solution for a number of reasons, so thank you. It can not come as a surprise to you that sometimes others don't appreciate the complexity of this kind of work. I have a long background in process automation and almost any associated process you can think of. Most people just aren't interested in what it takes to make the things they use everyday. I hope that gives you some insight and I will be willing to help in any way I can, although I'm not sure how that would be other than to try to break it. You could be surprised at the new tricks old dogs can learn though...

defazioa, 2023-12-15:
It works! Thank you, and you were right, it does exactly what I really wanted; the feedscrew will turn all the time at arbitrary speed. In move/sync mode the distance to go counts up, not toward zero, so it never comes to a stop. It's just a weird thing, I know I have something wrong with the microstep math and I was hoping to find clues there. Since I have a DRO I'll be able to tell how far I've traveled. I'll figure it out, for now I don't care. I'm very happy with the result, it's as quiet as a grave even at ridiculous speed, and that was paramount for me.

jschoch, 2023-12-15:
Just reverse the motor direction, should be a switch on the controller to
do it, or swap the encoder signals.

defazioa, 2023-12-15:
Why didn't I think of that?  Just kidding, the leadscrew reverses when the spindle does, the "distance to go" counter only counts in one direction and never resets or loads the distance to travel variable.

jschoch, 2023-12-20:
were you able to get this worked out?  Wondering what documentation we can create from your experience to smooth out the process.

defazioa, 2023-12-21:
I have some notes, I'll get back to you. It is working, although I'm still
unable to get the "distance to go" value to do anything but increase. I
have tried reversing, both at the drive and in the app, swapped encoder
leads and rise detect to fall detect and back again. Still broken, so I
can't speak to the bounce function and there is limited threading
functionality. The biggest problem for a near novice is there is a steep
learning curve for understanding what the thing wants when you build it.
It's all in there, but if you try to be cute and choose your own
directories when extracting you will likely get into path trouble you won't
understand. I know this is elementary stuff, but it's a stumbling block for
the masses. I'm not sure about the web interface, it's a mystery box if it
doesn't connect. The ESP32's are great and fast, but somewhat sensitive to
noise and require a level shifter which isn't especially friendly if you
aren't familiar. If there's a glitch and the radio doesn't start you're
left scratching your head. I'm committed to trying my feeble hand at using
this hall switch as an index pulse. This way you can sync to the spindle
every rotation and use the thread dial if you have one. Only speaking for
myself, I prefer at least a few controls I can put my hands on. It seems
best that even though the method of gearing is totally different at the
core, the end user experience remains as unchanged as is practical. But
this is a philosophical matter.

jschoch, 2023-12-21:
> I have some notes, I'll get back to you.

great, any notes would be appreciated

> It is working, although I'm still unable to get the "distance to go" value to do anything but increase. I have tried reversing, both at the drive and in the app, swapped encoder leads and rise detect to fall detect and back again. 

You may need to swap them both.  Here are a few things to try:

1. when you spin the spindle by hand anti-clockwise does the encoder counter value in the debug screen go up or down?  (you need to be out of startup mode to see this i belive)
2. If you spin the spindle in reverse does the distance to go counter decrement?
3. if you swap the encoder A/B wires does the counter do the opposite of what it did in test 1?

> Still broken, so I can't speak to the bounce function and there is limited threading functionality. The biggest problem for a near novice is there is a steep learning curve for understanding what [...]

did you hook up any LED's to the LED pins?  

>The ESP32's are great and fast, but somewhat sensitive to noise and require a level shifter which isn't especially friendly if you aren't familiar. If there's a glitch and the radio doesn't start you're left scratching your head.

my stepper controller doesn't need a level shifter, what model are you using?  Also I've read the pins are 5v tolerant (undocumented feature) and i think you could pull them up to 5v if it is the right kind of controller connect (open collector?) 

> I'm committed to trying my feeble hand at using this hall switch as an index pulse. This way you can sync to the spindle every rotation and use the thread dial if you have one. Only speaking for [...]

I'm not quite sure how that would work, if you use the threading dial and disengage the leadscrew you lose position.  this would be ok for "feed mode" but not for the moveSync modes.  Maybe worth opening another issue to discuss this as well as what would be needed for the encoder to use an index pulse.  Right now there is no logic to deal with that and we'd need to setup another interrupt.

jschoch, 2024-01-16:
closing this out now.

closed 2024-01-16 by jschoch