jschoch/ESPels #38: Math question

defazioa, 2024-01-01:
Hello again. I have been using your project for a couple weeks and I believe it works as you intended. I have a strange issue with the math with regard to feed per rev. When I dial up 25mm in the move sync mode with backlash out of the system, I only move 24.840. It's repeatable and linear so it stacks up over distance. The puzzling thing is that I thought if I was off it would be by a multiple of something or a reasonably round percentage, but it's this oddball number. There are some things I had wrong out of the gate; the encoder I was sent was 600 cpr not 360 as I thought, which was exacerbated by the fact that the encoder runs at 2:1 vs. the spindle due to gearing. I also failed to factor in the 4:1 ratio of the servo to the leadscrew, but these were the dumb errors I was expecting, not to come up .6% short. Any ideas? The only thing approaching a glitch I've found is that when you cancel a bounce move early, it still runs the entire return cycle at speed. This can cause some excitement and happens just when you start to trust the thing. I had a hell of a time with hardware this time as never before with the ESP32. I had some transients or similar causing flash corruption and hangs on boot. I have moved the pulse and direction pins to 22 and 23, there's some talk in the documentation about GPIO 12 causing serious misbehavior if pulled high during boot and causing this, and it seems better. I've mitigated noise as much as possible but there's a couple switching supplies sharing the enclosure which makes it a challenge. It's behaving now, I am getting used to the web interface but I wish the buttons were bigger. It is very responsive with a robust connection. Thanks again, it works a treat.

jschoch, 2024-01-01:
Somewhat hard to say without more details.  What are the exact settings you have setup; CPR, steps, and lead screw pitch?  It could be noise on the encoder signals.

I haven't touched the release version in quite a while (over a year) and so I can't rule out a bug causing the error.

I have a newer version that has a bunch of updates in it but there are some issues with the frontend and S3.  Perhaps I should polish it up a bit and get it ready to release if you want to test it out.  This may require you reflash/OTA update and to serve the html/javascript locally unless i can figure out how to fix the issue (CORS and SSE on S3).  The new version uses SSE to send updates and it sends them quite a bit faster vs sending everything over the websocket. Let me know if you want to try it out.

Regarding the buttons, I can also make some adjustments to the UI, the newer button has a bit more polish but i've not put all that much work into optimizing the button sizes, this is fairly simple to do though.

jschoch, 2024-01-01:
two questions: 

does the error change when you change the pitch but keep everything else the same?

does the error change when you run the spindle at different speeds?

defazioa, 2024-01-01:
The error is consistent, but I haven't done exhaustive testing. I solved the problem of the small buttons by mounting a tablet to the lathe. I'm very tempted to just do the math and put in whatever scalar value gets me to the number, but as I'm typing this it occurs to me that there are parameters like accel/decel in the integrated servo itself that could be causing it. I haven't really been able to properly tune the thing (JME model# IHSV57-30-18) because I am struggling to find the configuration software in English. I have a Mandarin version, but it's very cumbersome with Google translate and it's not the kind of thing you guess at. I am happy to try any versions of either the UI or the firmware prior to release if that helps. I have tried some variations here I found here among the issues dialog with varying degrees of success. I would like to talk about your original buttons when there's time to see if the functionality is still there and how one might go about using them.

digiexchris, 2024-01-03:
how is your encoder mounted to the spindle drivetrain? I've seen several optical encoders be very very picky about not having vibration, and I've found belt drives to be surprisingly troublesome in that regard if the alignment isn't perfect. I have a lathe that has a 4:1 reduction between the spindle and the first change gear as well, and I ended up driving directly off of the spindle using a pair of 100 tooth gears instead, the backlash in the 4:1 drivetrain to the encoder was causing some undetermined positioning errors, but that was more like a consistent loss of steps per pass so threads would double track, rather than a percentage movement error. Double tracking would be an interesting thing to double check in your case too.

 I would expect the error to change with pitch in that case though, like a twice as fine pitch should be double the error. Have you measured the leadscrew pitch over 100mm or so? If it's very consistent, I'd start by measuring the mechanical system. .6% won't be noticeable in a produced thread that needs to engage over the length of a nut, so your lathe manufacturer may have made a mistake there, or there is wear, or something else.

Might not be a bad idea to rig an LED to the in-position pin on your driver, and keep an eye on it during the start of a cut, I wonder if it lags a little per revolution of the leadscrew due to binding or something?

defazioa, 2024-01-04:
I'm glad you're watching, Chris. I feel like you accidentally got me started on this. I am going to do some real documented testing this evening and should have a coherent update to contribute. As far as the setup goes, I made a bomb proof mount for the encoder and took pains to align the belt drive properly so I don't suspect it. I have some experience with that stuff, I think you can see it in the pictures I posted in the other issue thread of my original set up. You're absolutely right about monitoring the position error, this thing is pretty slick, it will let you set a max position error, drive stiffness, accel, decel and a bunch of other stuff - and even graph it for you. The only thing is that I'm having a hell of a time finding a copy of the configuration software tool in English. After reading what I could of the documentation, it almost looks like this thing could do a rudimentary version of this on it's own if you were willing to do the math and enter it into a register for it to read. It's amazing how cheap this stuff is, when I saw a servomotor in industry for the first time a drive this size cost about half as much as a good used car, this one was less than $200 and is far more capable. I checked every possible mechanical cause since the error was so weird, there isn't much left to bind because the entire contents of the gearbox are now in a drawer. It may be a bit extreme, but I didn't want to have to hear those gears again unless I was willing to work for it.

jschoch, 2024-01-04:
I'm working on testing the newer version, @defazioa  can you provide your leadscrew pitch setting, micro steps, and spindle encoder resolution setting?

digiexchris, 2024-01-04:
I have dreams of making this work:

https://docs.broadcom.com/doc/AV02-2792EN

Absolute magnetic encoder. Would need to figure out how to radially align a magnetic field with the spindle though. Maybe a big enough DC motor magnet to wrap around the spindle...

jschoch, 2024-01-04:
@defazioa   I have merged in the changes over the last 6-9 months into the dev branch on this repo as well as the espelsfrontend repo.

This version uses SSE to send updates from the esp32 to the client, and while it is impressively more realtime, I can't quite figure out how to serve it via Amazon S3, so you'll need to run a local web server ( or maybe you can just load it from a file ) to test it until I find a eventsource polyfill that allows sending the correct preflight headers.  

Do you know how to checkout different branches with git?  

assuming you've cloned it and you have git installed you just do 
`git checkout dev`
in the root directory of the project

defazioa, 2024-01-05:
> I'm working on testing the newer version, @defazioa can you provide your leadscrew pitch setting, micro steps, and spindle encoder resolution setting?

The settings are:
Leadscrew pitch-4mm
Servo steps per rev-800
   (This is not the typical stepper chart I'm used to with native steps and a multiplier, it has dip switches which are set for 800. This value is used for native steps, additionally the belt drive is a four to one reduction so...)
Micro steps-4
Encoder pulses per spindle rev-4800
   (I physically verified this value because it was driving me nuts. It is a different resolution than I ordered and is not intuitively labeled.)
I'm sure I can manage the repo, if I need help I'll let you know. 
I had two events last night that I hadn't seen and I thought you might be interested in. At one point I was poking at something on the bench with the spindle running happily along when it grew a mind of its own and started creeping towards the chuck. I just happened to be looking at it and it was my fault for leaving the half nut closed. No harm done.  A different issue but under similar circumstances happened some time later when the leadscrew wouldn't change direction with the spindle when in feed mode. And in the wrong direction to boot. These gremlins seem to appear when you haven't used the UI for a while. It seems to misbehave if you ignore it. I'm looking forward to seeing what's new, thanks.  Oh, I tested the position error using my DRO and all in one direction. I cut at multiple speeds and feeds, under load and not, and the variation in the values was not more than 50 or 60 microns, but all over the map. Frankly, I don't know what it means if it means anything.  Thanks again, talk soon.

digiexchris, 2024-01-05:
> I very nearly did this after seeing it in a YouTube video, but I elected
> to go with the integrated motor. All you need is the chip and a
> diametrically magnetized magnet. They have both poles on the same face, and
> all you have to do is super glue it to the rear end of the motor shaft and
> mount the chip within 3mm. Cheap too.

I meant for the spindle encoder. In reality, if a stepper has enough torque
and the acceleration  values are reasonable, the closed loop feedback isn't
necessary (proven by CNC around the world every day). But an absolute
encoder on the spindle I think has merit. It can never get lost.

jschoch, 2024-01-05:
> @defazioa I have merged in the changes over the last 6-9 months into the dev branch on this repo as well as the espelsfrontend repo.
>
> This version uses SSE to send updates from the esp32 to the [...]

let me know if you can test the new dev branch per the above.  It has a bunch of fixes in it.

defazioa, 2024-01-09:
I compiled and flashed the dev, sorry it took until tonight. Circumstances. I am unsure how to connect to the front end, but I only tried briefly. I'll need some help with that.

jschoch, 2024-01-09:
You should be able to use the build file here: https://github.com/jschoch/espELSfrontend/raw/dev/build/index.html

You can serve this with any web server or on a PC just drag it into your browser, it should have all the dependencies.  Note the format for the esp32 websocket has changed, you just need to put in the IP address.

jschoch, 2024-01-11:
any luck with the frontend?

digiexchris, 2024-01-11:
if only we made money with this, so we could use a github action to build, request a cert, and host the FE :D I wonder if I still have that electron branch around that had an installable APK...

defazioa, 2024-01-13:
I'm just getting back to it. Brutal week. This is straying into foreign territory for me so please be patient.

defazioa, 2024-01-13:
Okay, man. I have it running on the tablet at the lathe with the correct IP, but it doesn't connect. I guess I'm doing it wrong, but here's what I know; I compiled and flashed the dev, I checked my router to make sure the lathe is on the client list, and I double checked that everything is the same as it was when it was functioning. Maybe I'm not your guy.

jschoch, 2024-01-13:
What does the serial output of the esp32 say?  
Can you ping the IP address?  
Can you take a screenshot of the network screen?

jschoch, 2024-01-13:
also, what tablet are you using, have you tried a laptop/desktop?  If you can run the frontend on a laptop/desktop can you open the browser developer tools and look a the console messages?  ( to open dev tools hit <ctrl> <shift> <i>  keys all at once)

defazioa, 2024-01-13:
> tools and look a the console messages? ( to open dev tools hit * keys all

I can ping the IP, it's alive. I can also see it on my managed router,
although even though I changed the name to lower case as instructed, it
shows up on the router in upper case as before. I suspect this is a quirk
of the router. I'm not out in the shop yet, but the serial output from the
esp said something about a bad nvconfig and that it didn' tlike the math
with zero pitch. No more running dB display. I opened the frontend first on
my lenovo laptop, I also made some attempts with my phone using Chrome and
the native browser for each. Chrome is generally unfriendly towards
unsecured ip's. The tablet is a Samsung Galaxy Tab  and is reasonably new,
but too big and a lot out of place where it's mounted. But that's out of
scope. I will try again this morning, I certainly hope my fortunes are
better. I want to test the thing and give you data, but in the interest of
beginning at the beginning, the installation has been difficult to say the
least.

jschoch, 2024-01-13:
can you reboot the esp32 while the serial is connected ( the reset button on the board) and paste in the output here?

defazioa, 2024-01-13:
Yes, I can. Give me a minute.

defazioa, 2024-01-13:
**here it is. I don't know why, but the steps are not what I put in the config.**
igsip: 0, SPrecalculated denominator: 19200
Set ELS Factor pitch: 0.100000 nom: 320 den: 19200
Setting up WiFi
reConnecting to Wi-Fi...
SSID: "" :p ""
..Connected to Wi-Fi.
Connected. IP=192.168.1.224
HTTP websocket server started
[   423][I][ArduinoOTA.cpp:141] begin(): OTA server at: esp32-24dcc39a40b8.local:3232
NV Config?
1
Loaded Configuration com version 0.0.6 lead screw pitch: 4.000000
Loaded this NvConfig doc{
  "i": 1,
  "lead_screw_pitch": 4,
  "spindle_encoder_resolution": 4800,
  "microsteps": 4,
  "EA": 25,
  "EB": 26,
  "ZP": 23,
  "ZD": 22,
  "motor_steps": 4000,
  "native_steps": 1000,
  "vsn": "0.0.3"
}
WS Log: Pitch 0, no good
Something wrong with this NvConfig!!!!!

WS Log: Pitch 0, no good
Using External PULLUP/PULLDOWN

Stepper timer starting

Stepper timer init complete
setup done
CPU: 240updating modes: 0

jschoch, 2024-01-13:
This all looks normal.  can you open the web browser's developer console and add here the output?

jschoch, 2024-01-13:
it may show "connecting" but if this icon in the upper left is green it is connected.

[image]

defazioa, 2024-01-13:
What do you need? I have it open, what are you looking for?

jschoch, 2024-01-13:
mine looks like this: 

[image]

defazioa, 2024-01-13:
I got it connected on the laptop, but I'm struggling with the tablet. I had it running, but now all I get is the full text and I can't remember how I got it to run. This is a pain in the ass.

jschoch, 2024-01-13:
>  but now all I get is the full text and I can't remember how I got it to run. 

i've no idea what you mean by "full text"

when run from the laptop does the inaccuracy go away?  that was the whole reason you wanted to test this new version.  If that is fixed I can try to clean up the connection code a bit as it is a bit confusing and it isn't remembering the IP address in a cookie like it should.  I also want to fix the SSE issue if possible.

Regarding the tablet, have you tried to install chrome or firefox?  Is it using some proprietary browser like samsung's or something?  

How are you serving the index.html?  

Have you tried your phone?

defazioa, 2024-01-13:
I got it talking. You are aware that you have to re-enter the network each time you open it I assume. Feed mode does not indicate it is turned on.

defazioa, 2024-01-13:
There is no RPM info, and it will not start in move sync unless you e-stop feed mode. I ran a test at 10 mm and got 9.900. 25 returns 24.805. These values are with backlash removed.

defazioa, 2024-01-13:
It's better, no more signed values for move distance. It also seems faster and hasn't changed direction on me suddenly. The errors are smaller but still present.

defazioa, 2024-01-13:
I take that back. I just went back to feed mode and at first it didn't run, then it decided to run backwards. It could have already been trying to run in another mode but one can't tell without it saying so. Best practice to release the half nut if you have to walk away and double check which way the screw is turning before you engage the feed.

jschoch, 2024-01-13:
"lead_screw_pitch": 4,
"spindle_encoder_resolution": 4800,
"microsteps": 4,
"motor_steps": 4000,
"native_steps": 1000,

these settings are very confusing, if your microstepping is 4 that should mean that it takes 800 steps to turn the motor 1 full revolution.  so this means "motor_steps" should be "native steps" * "microsteps", but yours is 4000.  Did you manually change native steps?  

Also, the spindle_encoder_resolution set to 4800 is odd since I think you said it was something more like 600.  The spindle_encoder_resolution should be the counts per revolution, and this should be 4 * pulses per revolution of your encoder (using quadrature encoding).  4*600 is 2400 which is 1/2 what you have configured.  

Finally, the lead screw pitch should be based on the distance the leadscrew travel when your motor turns one revolution.  If it is directly connected that would just be the measured pitch of the leadscrew.  If it is through a gear then you have to modify the measured pitch of the leadscrew by the gearing ratio.  So for example, my lathe has a 2mm pitch lead screw but my motor is connected via a belt that has a 3:2 reduction.  this results in a 1.333333333333 (1 1/3) mm leadscrew pitch configured in the ELS.

So i'm guessing that while you are close (maybe by accident) you have something not quite right and it is most likely the lead screw pitch.  From Weiss it says the leadscrew is imperial and so it would be odd if it was a whole number.  Do you know the exact gear ratios between the servo and the leadscrew?  Have you measured the leadscrew pitch precisely and ensured it is metric vs imperial?

I'm very troubled ty it "running backwards", but i'm not sure how to troubleshoot or reproduce the error.  Please try to note if this is a first run/first boot issue or if there is a way to repeat the error.  It may also happen when transitioning from a particular mode to another after a certain startup or stop state or state transition.

defazioa, 2024-01-13:
If you look back in this conversation you will see the data you've asked
for. The motor runs a 4:1 reduction to the leadscrew, so I use a value of 4
for microsteps. The dip switches on the motor are set for 800 steps per rev
so I use that value as native steps. The encoder is 600 per and is geared
to turn twice for every spindle revolution. 2 x 600=1200, multiply by four
and you get 4800. It seemed high to me so I counted it, it is indeed 4800.

defazioa, 2024-01-13:
The direction thing seems to happen when changing modes or leaving the
display idle for a while. It will require an e-stop to recover. This hasn't
changed.

jschoch, 2024-01-13:
> The encoder is 600 per and is geared
to turn twice for every spindle revolution. 2 x 600=1200, multiply by four
and you get 4800. It seemed high to me so I counted it, it is indeed 4800.

this checks out.

>The motor runs a 4:1 reduction to the leadscrew, 

if you put "4" in for lead screw pitch you are telling it the pitch is 4mm even after  the motor reduction.  I believe the pitch can only be set in mm.  Can you measure the actual pitch on the leadscrew in mm (even though it is likely in inches)?  Again, if your "WBL250F" lathe is imperial it shouldn't be a whole number.  You should enter the measured value in mm and divide it by 4 (your motor reduction) and enter that as the lead screw pitch.

>  so I use a value of 4 for microsteps. The dip switches on the motor are set for 800 steps per rev
so I use that value as native steps. 

steppers come in 200 and 400 native steps.  servo's have no notion of native steps.  your servo can act like a standard servo and so if you set the steps per revolution to 800 you are simulating 4x microstepping with a 200 step stepper or 2x microstepping with a 400 step stepper.  all that is somewhat moot, the total steps per revolution is all that should matter.

Native steps should be 200 and microsteps 4.  200*4 = 800.  

Below the value "Motor steps per revolution" should be whatever you set on your servo

[image]

Your is set to 4000 steps per revolution and i'm guessing that you lucked out and somehow the leadscrew value of "4mm pitch" you entered is very close and somehow it all works out close but it is not correct.

So we should nail down the pitch of the leadscrew and then update the setting to match the reduction.  then get the steps correct to match your servo's setting.

jschoch, 2024-01-14:
I put the number in real quick assuming a 10TPI leadscrew and it does come out very close and I'm guessing this is your issue.  I have scales on my machine and it repeats +/- 0.01mm with the backlash removed.

[image]

So again, per the more detailed post directly above this we need to verify the leadscrew is 10TPI, and if it is your "leadscrew pitch" in the ELS should be set to 0.635  (this has to be in mm AFAIR).  Then the total "motor steps" should match your servo's setting of 800, you can do this by resetting the native steps to 200 and leaving the microstepping "4".

defazioa, 2024-01-14:
You are absolutely right. I have a spec sheet here that says only the cross feed screw is SAE. Wrong. It's 8 tpi. I was able to change it but the motor steps are still screwed up, I probably was cooking the books a little the last time just to see how close I could get. The problem is, I've flashed the new code with the correct config twice now, once over USB and once OTA, and the nvconfig doesn't change. I think if I can get the motor steps per rev to change to 3200 (800 native x 4 micro) I think we're golden I was able to recreate the spurious movement thing; I walked away (with the half nut open) for about 45 minutes with the unit still powered, and when I returned the move synced with spindle I had used before I left, now would never reach it's terminus. There is no option to terminate move except the e-stop, I had already released the nut but I pressed it anyway. It stopped, but every move after that was in the opposite direction and never stopped at the appointed place. I had to power it down

jschoch, 2024-01-14:
> I think we're golden I was able to recreate the spurious movement thing; I walked away (with the half nut open) for about 45 minutes with the unit still powered, and when I returned the move synced with spindle I had used before I left, now would never reach it's terminus.

Can you create a new issue for this?  I'm still not sure what modes it was in or what exactly happened.

> I think if I can get the motor steps per rev to change to 3200 (800 native x 4 micro) I think we're golden

We should fix this properly.  Have you tried "reset config to defaults"?    This should reset the NV Config and recalculate the "motor steps" which is (microsteps * native_steps).  Assuming you are using the default 200 native_steps in your config 

```
-D Z_NATIVE_STEPS_PER_REV=200
```

This should now let you get to 800 (200 * 4) motor_steps.  We can confirm via the webUI or serial.

Maybe I should have it so you just enter the total steps per revolution and forget about native_steps and microsteps...

jschoch, 2024-01-14:
the reset button is here, please see the above response for the way to fix the steps.

[image]

defazioa, 2024-01-14:
I thought about trying the reset button, but I didn't know what was lurking
behind it. I'll do that this morning. I can change the native steps to 200
(which I agree is correct) if you want me to, but where do I bake in the
4:1 reduction? It seems like all that should matter is how many total steps
the controller has to deliver for one rotation. I also was starting to
wonder whether having the smart motor without integration could hurt you.
There's the possibility of them competing maybe. For wider release, if you
give well written instruction on how to calculate that number, the end user
can just enter that. Provided there's no magical math behind the scenes.

defazioa, 2024-01-14:
I'll open another issue for the direction issue and general schizophrenic behavior of my UI. I've been doing some reading and it seems there's a consensus that the flash of the ESP32 is somewhat delicate, I'd like to know if you have an opinion about that based on experience. Once I've flashed this thing a dozen more times I'll use a fresh controller to make sure there's no corruption.

jschoch, 2024-01-14:
> I can change the native steps to 200
(which I agree is correct) if you want me to, but where do I bake in the
4:1 reduction?

The reduction is accounted for when doing the lead screw pitch calculation.

So, the value for the leadscrew pitch in the els should be 0.79375, which is   

8TPI (25.4/8 = 3.175mm) divided by your gear reduction (4).

I need to update the docs for geared leadscrews so this is more clear.

defazioa, 2024-01-14:
I guess that's what I mean. The commutative property of multiplication means the result is the same, the difference is what makes sense to you. It's subjective. However you decide to do it, it's best to eliminate the human factor to the greatest extent possible in my opinion.

jschoch, 2024-01-14:
best to follow convention so that if something changes you don't get stuck with bad assumptions.

see below as to how close the ratio turned out to be, is is the ratio for driving a 0.1mm pitch per Z rotation.  this is your 6% error.

correct value  0.02099737533 
lucky but wrong   0.02083333333

defazioa, 2024-01-14:
I changed the values to:
pitch - 3.175
native steps - 800
microsteps - 4
There is no field for motor steps per rev any more.
With these values it repeated within 10 microns. That's well within the margin of error for this turd. I can try it with the ratio calculated into the pitch, these were the values it came back with when I hit the magic reset button. Unfortunately, after getting the measurement, now it doesn't stop in either move sync mode. Just goes on forever like feed mode. You removed the move cancel button so this is a fatal error. I'll detail it in another issue.

jschoch, 2024-01-14:
the pitch should be 0.79375
motor steps per rev is calculated, you can't enter it as a value directly.
you have to set native steps to 200 and microsteps to 4, that should net you 800 motor steps per rev (as calculated)

While your current setting may work It would be better to follow the convention because I can't assume anyone would configure it like you have.

You should also be able to change the leadscrew pitch when you change the gears if you wanted more torque for example, but I don't think it is quite as clean the way you are doing it now unless the gear ratio happens to be a whole number and not something like 1/3.  Also changing native_steps requires a reflash and changing leadscrew_pitch is just done in the webUI.

> Once I've flashed this thing a dozen more times

You should be able to flash thousands of times, just make sure there is good usb cable and you are not in an electrically noisy environment (e.g high power brushed motors running next to it).  where have you read you need to get a "fresh" esp32 after only a few flashes?

jschoch, 2024-01-18:
seems noise related, lots of fixes implemented.  closing for now.

closed 2024-01-18 by jschoch