jschoch/ESPels #39: CORS and SSE

jschoch, 2024-01-03:
I did some research on the CORS issue again, here's the state of things.

CORS-RFC1918 prevents "more private" network connections without a  preflight " Access-Control-Allow-Private-Network: true" header sent from the browser and acked from the esp32.  There is no issue if you are serving the static html from your private network.  Adding the header on the esp32 is fairly easy to do but....

EventSource API does not allow you to add headers and...

Chrome/Firefox don't issue the preflight request because they don't seem to see the rfc1918 address, but they do kill the connection because they see a rfc1918 address....

So, it appears I need to re-write the frontend to use a polyfill that allows for event source preflight headers to be sent via the fetch() api.

Before I do that I plan on taking the current experimental branch and getting them released which should allow for configurable SSE or Websockets for updates, as well as options for  interrupt driven/timer driven control logic.

jschoch, 2024-01-16:
Firefox on PC and Firefox Beta in google play store are both working with the S3 link

jschoch, 2024-01-18:
Use turbo links or something like suspense/lazy in react to lazy load the JS from a stub embedded in the esp32 and make it somewhat easy to update the target source.

defazioa, 2024-01-19:
There is a minor issue I have discovered with the UI. After running for about an hour, the UI reverts to startup mode without explanation. It is not multi client related, I have checked that no other devices are running it aside from the tablet. I discovered this in addition to another potentially serious bug; if you inadvertently double press the start button on the UI, it will complete the first distance, almost imperceptibly stop, then do the same distance again. There is no indication that this will happen of course. The distance to go display is coming up now but intermittently freezes on one value. Unlike similar conditions before, travel continues and then stops at the prescribed place. I assume most of these concerns are communication based so I put them here. If they belong elsewhere, please put them in the appropriate place.

defazioa, 2024-01-20:
What are your UI plans for a wider release? I suppose my question is, will it be possible to port this over to a local display using wired communication, or possibly a second ESP to share the comms and processing burden and communicate locally over ESPnow or similar. I assume that once the project is complete in your eyes, the need for OTA development updates will be greatly diminished and possibly outweighed requests for a wired solution. I am not questioning your vision, just asking a question. I just saw a tft display using an ESP for the controller, and they even broke out a few IOs for you to use. That's interesting.

jschoch, 2024-01-21:
> There is a minor issue I have discovered with the UI. After running for about an hour, the UI reverts to startup mode without explanation. It is not multi client related, I have checked that no other devices are running it aside from the tablet.

If I were to guess this is likely because the ESP32 rebooted or crashed.  See if you can see a crash log in the serial monitor next time it does this.

> I discovered this in addition to another potentially serious bug; if you inadvertently double press the start button on the UI, it will complete the first distance, almost imperceptibly stop, then do the same distance again. 

I don't think I have a start button.  please screenshot and explain in more detail exactly which mode this happens in and exactly what was clicked and what all the move settings were.

> There is no indication that this will happen of course. The distance to go display is coming up now but intermittently freezes on one value. Unlike similar conditions before, travel continues and [...]

This is a bug if it is possible.  I plan to re-implment the finite state machine for moveSync which should prevent this from happening.