jschoch/ESPels #41: Unexpected behavior in Move/Sync mode

defazioa, 2024-01-14:
While doing testing, I used the move synced to spindle mode in concert with the DRO to resolve a ratio problem. It worked as expected, but when I tried to run a bounce move immediately after, the move continued past the expected return point and continued running after dis-engaging the half nut to avoid a crash. The move cancel button has been removed so the only way to stop the control is by emergency stop, but this doesn't resolve the problem when starting back up. The controller has to be reset. I will attempt to flash the project on a new controller to reduce the likelihood of a hardware problem.

jschoch, 2024-01-14:
I think that both the "forgetting" of the espELS IP and this are related to serving the frontend via a file vs serving it via a webserver.  It seems that cookies don't work when serving from a file and when I tested bounce mode from the file it also didn't update the UI.  I'm digging in to figure out what the problem is.  I think I saw this with my tablet, you could try to tell the tablet to use "desktop mode" vs "mobile mode".

[image]

if you have python 3 installed you can serve the index.html from your PC or laptop like so:  

`python -m http.server 8080`

or if it is a windows binary 

`python -m http.server  <what ever port number you want>`

then you can use http://[laptop ip address or name]:8080/index.html

I'm working on getting this so it can run from Amazon AWS S3 again (like you used with the main branch version) but there is quite a lot to do to make that work correctly due to the CORS RFC1918 SSE issues.

It really doesn't matter what the webserver is, if you have node.js and react setup (requires version node.js  v16.20.0) then the dev server works fine.  I should probably also update node.js

defazioa, 2024-01-14:
I am more concerned with the absence of a cancel button and it's attitude when trying to recover. The distance to go was nice too. I don't know how you have it set up, but the first thing it should do when you make a selection for anything is to cancel whatever was chosen before. Now that I have it talking I haven't had any more trouble in that respect. The biggest headaches with the UI are those I outlined above and having to reenter the network address after restart. I will say that I didn’t expect it, but I have trouble trusting it. I'm on high alert while it's in use but I guess that's not a bad thing.

defazioa, 2024-01-15:
I flashed a new controller and installed it and it hasn't gone the wrong direction or done anything scary since. Probably a coincidence, there's also that "Mandela effect" thing that makes one believe that things settle in and perform better over time. I haven't tried chasing a thread yet to see how it lands, I'll set that up now. Do you have plans for multi start? I don't have a use for it, but I thought it could be an interesting way to knurl.

jschoch, 2024-01-15:
> I am more concerned with the absence of a cancel button and it's attitude when trying to recover. The distance to go was nice too. I don't know how you have it set up, but the first thing it should [...]

were you able to serve the index.html file via a webserver?  If so can you confirm the cancel button is back?  Again, i think this is somehow related to serving it as a local file vs serving it via a webserver.

Tomorrow I'll work on trying to get SSE to work via AWS S3.  this should solve a bunch of the problems.

defazioa, 2024-01-15:
Yes, I am serving it that way, not as a local file. I understand only part of what the trouble is and I'll leave that in your hands. It has not done anything unexpected for a long while now. I also think I wasn't really treating it very gently when changing modes or starting and stopping. I probably exacerbated the issue. I think I said this, but the RPM doesn't display. I don't think the DRO is functional either in this version, I can't remember because I didn't use them. An angle display for the spindle might be more useful. And giant "GO" buttons in all the places you start moves. I'm so out of touch I though SSE was one way.

jschoch, 2024-01-15:
> at way, not as a local file. I understand only part of what the trouble is and I'll leave that in your hands. It has not done anything unexpected for a long while now. I also think I wasn't really [...]

The RPM, DRO, Angle, and buttons for canceling a move_sync move or bounce should all work.  I made some small tweaks, can you try this version?

https://raw.githubusercontent.com/jschoch/espELSfrontend/dev/build/index.html

Is the webserver on the same network as the espELS?

With the version linked above you should be able to more easily see if the SSE is connected or if has fallen back to using websockets for updates.

[image]

jschoch, 2024-01-15:
I think I found the issue!  It seems that having more than one client connected (browser tab, or mobile device like a table or phone) breaks the SSE updates and so the RPM, DRO, move cancel buttons all break.  Solution right now is to close all the other open browsers connected the espels.  Longer term this is something I think I can fix, but need to figure out exactly what the problem is.  Websockets dont' seem to have this same behavior.

defazioa, 2024-01-15:
Did you implement your virtual compound for threading? If you didn't I still have a problem, if you did I need to know how much it steps over so I can get the thread cut before the tool drifts out of the thread window. I sure hope it's the latter. That checks out on my end about multiple devices, I have it open on both the tablet and the laptop.

jschoch, 2024-01-15:
There isn't virtual compound in this version, I need to redo it.

jschoch, 2024-01-15:
can you let me know what browser you are using?  I have SSE on AWS working with firefox but not chrome....

Let me know if you want to test it with firefox.

defazioa, 2024-01-15:
I'll install Firefox and try it. Do I need a link? I was serving it from
the laptop with Serve. Would that have caused the same anomalies? I only
half know what I'm doing.

jschoch, 2024-01-16:
If you are running good and don't want to be bothered then you can just keep it as is.  Please confirm the DRO and move cancel buttons are working and let me know if there are any other issues.

Firefox should let you use the S3 link.  You'll also need to grab the latest version of dev.  If you used `git clone` to download my espELS repo (which you should do you don't have to download the whole thing every time) you can just run `git pull` and it will just pull down the changes.  It is in the dev branch.  This may not be required, and you can easily just try the link with firefox to see if you need to recompile and flash.

the link is: http://espels.s3.us-west-2.amazonaws.com/dev0_0_5/index-cors3.html
Make 100% sure you're browser isn't using https or you'll just get a blank screen and the page title (tab title) likely says "http not https"

Also, you may have to go back into config/settings and make sure Imperial is set for the units (assuming you are using imperial)

I've not tested with a tablet or phone yet

defazioa, 2024-01-16:
Imperial is dead to me. I'm working on it now. What time zone are you in?

jschoch, 2024-01-16:
I'm PST.

if you use discord that may be somewhat easier to communicate.

You will need to recompile/flash.  I remembered that I had to respond to HTTP_OPTIONS with  200 in the firmware for the CORS preflight check to work.

Also, I just checked and "firefox beta" in the google play store using the AWS S3 link above is working on my phone!!!!!

Another tip, you can append `?ip=192.168.1.87`  at the end of the url and it will plug that in to connect.  so in my case the full URL is `http://espels.s3.us-west-2.amazonaws.com/dev0_0_5/index-cors3.html?ip=192.168.1.87`

`

jschoch, 2024-01-16:
are you on discord?  My username is `schoch_`

Are you sure you compiled and flashed the dev branch?  If you are using git, do `git status` it should tell you the current branch that is checked out.

it should show somethiing like 
```
git status
On branch dev
Your branch is up to date with 'origin/dev'.
```

jschoch, 2024-01-16:
did you get it fixed?  It should be version 0.0.6 on both sides.

Console output (on the web side) and serial output (for the esp32) are super helpful

also, you may want to setup one of your spare esp32's on your desk so you can do quick tests like this.  You should be able to flash it and connect without it connected to the encoder or stepper.  may save you some trips to your lathe.

defazioa, 2024-01-16:
I'll flash an extra one. I finally got the react to work but I still don't have RPM. The difference is that it synced before, or it pretended to. The threads didn't track exactly now that I'm thinking of it. Sorry, there's question I haven't answered. I am not on Discord, haven't gone there and I'm somewhat afraid I won't get what's going on there. Move cancel button is back and works, even though it never moves.

defazioa, 2024-01-16:
Okay, I cloned it from the dev branch just now and flashed it, but it says "branch master" when I check status. I got all excited, but I have failed. The other instances of the repo have been deleted. What did I do?

jschoch, 2024-01-16:
> Okay, I cloned it from the dev branch just now and flashed it, but it says "branch master" when I check status. I got all excited, but I have failed. The other instances of the repo have been deleted. What did I do?

ok, to checkout the dev branch run this command

`git checkout dev`

then run `git status` and check to be sure you are on "dev"

then compile and flash

> I'll flash an extra one. I finally got the react to work but I still don't have RPM. The difference is that it synced before, or it pretended to. 

not sure what you mean, git sync'ed?   `git pull` should be the command to fetch the changes from github

>The threads didn't track exactly now that I'm thinking of it. 

Not sure what you mean, the thread pitch is off?

>Sorry, there's question I haven't answered. I am not on Discord, haven't gone there and I'm somewhat afraid I won't get what's going on there. 

It is just a chat server app, like IRC.  You can chat in realtime.  If you join you should be able to direct msg me or join a server like linuxcnc discord, or whatever topic there is likely a server for.

>Move cancel button is back and works, even though it never moves.

I don't know what this means.  can you take a screen shot?  here's how to use the built in screen shot https://support.microsoft.com/en-us/windows/use-snipping-tool-to-capture-screenshots-00246869-1843-655f-f220-97299b865f6b

jschoch, 2024-01-16:
Ok, here is the full list of commands for simplicity.  you should run it from the folders you cloned, if you put the cloned repo oin another directory from the zip/downloaded one you need your config.ini that you customized, put it in the project root where the platform.ini is.   git shouldn't overwrite this even if you checkout another branch.

```
git pull
git checkout dev
git status
```

I also just made an espELS discord server, so if you want to troubleshoot in real time join me there.
https://discord.gg/qseXnuTS

defazioa, 2024-01-16:
Thank you. Give me one second. I'm all over it. I was registering for Discord.

defazioa, 2024-01-16:
I had to change the "configured" variable in mine to "CONFIG_CONFIGURED" but that let it build. I flashed it but we're back to bad version on the tablet.

closed 2024-01-16 by jschoch

jschoch, 2024-01-17:
@deffazioa  I figured out why you can move async jog but not movesync.

When you moved your config you somehow used an old config which was missing a define.    check you config.ini and I think you will find that the values MODE_MOTION_INTERRUPT or MOTION_MODE_TIMER  are not in there.

In the most recent commit to the dev branch i've updated it so that you can't miss this anymore.  this was causing all the issues with no pulses being generated.

I've also added some debug info and fixed a few other bugs I found.  Please use this new react S3 Link.

`http://espels.s3.us-west-2.amazonaws.com/dev_06/index-cors5.html`

you need to run 

```
git pull
git status   
```
make sure you are on dev, if not run `git checkout dev`

then fix your config, and compile and flash.

Finally use the url above with Firefox to test.  ask on discord if you hit any snags

jschoch, 2024-01-17:
Please read the full previous update, then back here ( assuming you are reading via email)

in your config.ini you need to define MOTION_MODE 

Here is the stuff to add, use mode 1

```
####   Timer mode is not stable and interrupt mode should be preferred
    #       MOTION_MODE = 1  is for using interrupts to drive the motion calculations
    #       MOTION_MODE = 2  is for using a hardware timer and polling to drive the motion calculations
    -D MOTION_MODE=1
```