jschoch/ESPels #48: add state transitions and timer back to moveSync

jschoch, 2024-01-18:
this allows you to log when you move from "idle" to "running" to "done" and log it in the serial console.  Hard to do this without the FSM due to interrupt code.

defazioa, 2024-01-20:
The UI reverts to Startup mode after completeing a Bounce or Move Synced to Spindle move. I don't think it was this way when I started running it. I am running the same version on the same browser. It is consistent in Bounce, but it sometimes stays on the Move Synced screen before it reverts to Startup.

jschoch, 2024-01-21:
ya, i think this is because I use the same code for halt as I do for finishing the bounce.  should be a fairly easy fix  when I have time.