jschoch/ESPels #50: Compile Issues/problems

ronehorn, 2024-08-20:
1)UndefinedEnvPlatformError: Please specify platform for 'ota' environment
2)[{
	"resource": "/c:/Users/roneh/OneDrive/Documents/PlatformIO/Projects/ESPels-master (4)/ESPels-master/src/src/led.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'wifi_led' was not declared in this scope",
	"startLineNumber": 23,
	"startColumn": 22,
	"endLineNumber": 23,
	"endColumn": 22
3) [{
	"resource": "/c:/Users/roneh/OneDrive/Documents/PlatformIO/Projects/ESPels-master (4)/ESPels-master/platformio.ini",
	"owner": "PlatformIO",
	"severity": 4,
	"message": "`check_patterns` configuration option in section [env:cppchecker] is deprecated and will be removed in the next release! Please use `check_src_filters` instead",
	"startLineNumber": 1,
	"startColumn": 1,
	"endLineNumber": 1,
	"endColumn": 1
},{
	"resource": "/c:/Users/roneh/OneDrive/Documents/PlatformIO/Projects/ESPels-master (4)/ESPels-master/platformio.ini",
	"owner": "PlatformIO",
	"severity": 4,
	"message": "`check_patterns` configuration option in section [env:ctd] is deprecated and will be removed in the next release! Please use `check_src_filters` instead",
	"startLineNumber": 1,
	"startColumn": 1,
	"endLineNumber": 1,
	"endColumn": 1
}]

jschoch, 2024-09-16:
did you get this figured out?  Sorry for the delay in responding.

ronehorn, 2024-09-16:
No, not yet.

closed 2025-01-30 by jschoch