jschoch/ESPels #52: build error

RiickDEV, 2025-05-05:
I'd like to know why these errors are showing up
I created the config.ini file, but they still keep appearing.

[image]

-----------

[image]

jschoch, 2025-05-06:
what branch?

jschoch, 2025-05-06:
also your platform.ini, does it have this in it 

```
[platformio]
extra_configs = 
  config.ini
default_envs = 
	ESPelsTest
```

RiickDEV, 2025-05-08:
I found the problem  I had never used PlatformIO before, and it was uploading to all the environments listed in the platformio.ini. In Visual Studio, I selected the ESPels environment and the upload worked perfectly.

RiickDEV, 2025-05-08:
Which environment should I upload to? ESPels or ESPelsTest?

[image]
For anyone who has the same doubt I did, this is where I selected the environment.

RiickDEV, 2025-05-09:
Got it, I'll keep using ESPels
Thanks

closed 2025-05-09 by RiickDEV